EFeru/hoverboard-sideboard-hack-STM #2: is that same board

bekirbostanci, 2020-06-22:
hi 
I bought a hoverboard and mainboard is not looks like same but i am not sure. Can I use this ? 
[image]
[image]

KaanDemirkoparan, 2020-06-22:
Yes it is same. You can use it. 

You shouldn't use this repositorie's issue page for mainboard issues. It would be better to use (https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues) this page for issues or questions about mainboard.

bekirbostanci, 2020-06-22:
it is not working. st-link did'nt detect the board. 
thank you for your help.

closed 2020-06-22 by bekirbostanci