EFeru/hoverboard-sideboard-hack-STM #3: Schematics seem to have inverted polarity for +12v input

Poupoudoum, 2020-11-18:
Hi,

I've just exploded a 5V regulator and possibly a full sideboard while trying to test my very basic firmware : I only conencted +12V according to the schematics in the README.md.
On my board ground is on the gyroscope side and +12V on the ARM proc side.

(By the way mine is a GD32 while having the STM32 card layout...)

Thanks for your job !

EmanuelFeru, 2020-11-18:
Sorry to hear. The wires on these boards are tricky. They have to be checked. Sometimes even the wire colors do not match.
So your MCU is GD but the board layout is the STM version. Does the pinout match with the STM one or is completely different from both?

Poupoudoum, 2020-11-19:
My layout is the same as the STM variant.
It as some very minor differences, such as an unused second port for blue leds and the leds resistors are not of the same value.

Here is a picture of my board, with the mainboard connector plugged so you can check the polarity issue.

[image]

Poupoudoum, 2020-11-22:
You were right... My motherboard AND my sideboard have inverted connectors. so my red wire is ground and my black wire is +12v :-O
Thanks for this advice I checked this yesterday and it saved my board from another inverted polarity. I close this issue as it's not related to this hack but to my boards.

closed 2020-11-22 by Poupoudoum