EFeru/hoverboard-sideboard-hack-STM #8: dual inputs Serial + Sideboard not working

Tjaart-Hobby, 2021-12-16:
Hi
I am trying to get dual inputs working > [https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/242](url)
So now I am at "Serial + Side board "attempt:

1 Checked wiring polarity and unlocked with Cubeprogrammer 
2 Flashed VARIANT_HOVERCAR to side-board (to use RC receiver and later incorporate mpu for acceleration)
3 Flashed main-board with VARIANT USART:

This is gonfig.h on main-board

`#ifdef VARIANT_USART
   #define SIDEBOARD_SERIAL_USART2 0
  //#define CONTROL_SERIAL_USART2  0    // left sensor board cable, disable if ADC....
  //#define FEEDBACK_SERIAL_USART2      // left sensor board cable, disable if ADC or PPM is used! tk

  // #define SIDEBOARD_SERIAL_USART3 0
   #define CONTROL_SERIAL_USART3  1    // right sensor board cable. Number indicates priority.......
   #define FEEDBACK_SERIAL_USART3      // right sensor board cable, disable if I2C (nunchuk or lcd) is used!

  #define DUAL_INPUTS                 //  UART*(Primary) + SIDEBOARD(Auxiliary). Uncomment this to use Dual-inputs
  #define PRI_INPUT1             3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #define PRI_INPUT2             3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #ifdef DUAL_INPUTS
    #define FLASH_WRITE_KEY      0x1102  // Flash memory writing key. Change this k.........
    #define SIDEBOARD_SERIAL_USART2 0   // left sideboard
    //#define SIDEBOARD_SERIAL_USART3 1   // right sideboard
    //#define CONTROL_SERIAL_USART3 1  //tk added 
    #define AUX_INPUT1           3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT2           3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #else
    #define FLASH_WRITE_KEY      0x1002  // Flash memory writing key. Change thi.........
  #endif

  // #define SUPPORT_BUTTONS_LEFT       // use left sensor board cable for button inputs.  Disable DEBUG_SERIAL_USART2!
  // #define SUPPORT_BUTTONS_RIGHT      // use right sensor board cable for button inputs. Disable DEBUG_SERIAL_USART3!
#endif
// ######################## END OF VARIANT_USART SETTINGS #########################`

All I am getting is Serial on right cable > working(tested with Arduino)
                         but side-board left cable nothing(don't know how to check side-board function)
[image]
Note: STM32F103C8T6 

Connected RC receiver : picked 5v and ground from board and USARTT1_RX connected straight to IBUS on receiver
Whats the recommended resistor for?

EFeru, 2021-12-16:
I need to check but should normally work because for Hovercar I use ADC + Sideboard.

Can you temporarily try Hovercar with 2 Potis, sideboard and RC? Then if that works switch to serial instead of Potis.

Tjaart-Hobby, 2021-12-17:
Hi
I got ADC connected to left cable and sideboard to right cable. ADC seems to work although only forward and turn one way. I am sure I could work out why that is but I am only using it as a test tool (just like the arduino).
The sideboard doesn't seem to work. There is a slight change in the "noise" when I switch the transmitter on and off. <noise from PWM to mosfeds ,I think>
link to vid:[https://www.youtube.com/watch?v=pI3ASJRo9rw](url)
My gonfig.h (main board) + sideboard flashed with hovercar

`  #define CONTROL_ADC           1         // use ADC as input. Number indicates priority for d.....

  #define DUAL_INPUTS                     //  ADC*(Primary) + UART(Auxiliary). Uncomment this to use Dual-inputs
  #define PRI_INPUT1            3, 0, 0, 4095, 0      // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #define PRI_INPUT2            3, 0, 0, 4095, 0      // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #ifdef DUAL_INPUTS
    #define FLASH_WRITE_KEY     0x1101    // Flash memory writing ke......
    #define SIDEBOARD_SERIAL_USART3 0
    //#define CONTROL_SERIAL_USART3 1       // right sensor board cable. Number indicates priority....
    //#define FEEDBACK_SERIAL_USART3        // right sensor board cable, disable if I2C (nunchuk or lcd) is used!
    #define AUX_INPUT1          3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT2          3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #else
    #define FLASH_WRITE_KEY     0x1001    // Flash memory writing key. Cha....
    #define DEBUG_SERIAL_USART3           // right sensor board cable, disable if I2C (nunchuk or lcd) is used!
  #endif
`
I also left SIDEBOARD_SERIAL_USART3 commented out with CONTROL_SERIAL and FEEDBACK_SERIAL defined > did not work.

Tjaart-Hobby, 2021-12-17:
I re-flased the sideboard with IBUS channels set to 6 from 14. 
In the process I lost connection from the sideboard to mainboard (wire fell out) and the main board started steady beep. Like it does when no input detected. I conclude that since it doesn't beep when connected that the sideboard works.
Changing the IBUS channels did not help.

My side board is somewhat different than yours. The recommended 100 ohm resister: Is that used to limit current in RX pin or as a pull-up or pull-down resistor?

EFeru, 2021-12-17:
Resistor is just to limit current. In the wiki of Hovercar is described which channels are needed.

For the remote to work you need to flip switch SWA to RC position
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR

No beep means indeed the sideboard communicating with mainboard.

Tjaart-Hobby, 2021-12-17:
I didn't see the wiki, sorry. I wasn't interested in that variant. I will go through it when I am back home. Travelling in city now.

I did flip the switches though and did not work. The specification leaflet I got with my transmitter shows 6 channels.
If switches are on 7+ channels its probably my issue.

Tjaart-Hobby, 2021-12-17:
confirmed transmitter only 2 aux channels, 5+6. looking at code snipped could I just amend 

'cmdSwitch   = (uint16_t)(switch_check(ibus_captured_value[6],0)      |      // Channel 7'
to:
`cmdSwitch   = (uint16_t)(switch_check(ibus_captured_value[4],0)      |      // Channel 5 ` which should be channel 5?
going to try it any way tomorrow unless I hear back from you before then. ;)

EFeru, 2021-12-17:
Yes, that should work. It is indeed the changes needed if your transmitter uses different channels.

Tjaart-Hobby, 2021-12-18:
Still does not work. I also changed IBUS channels in config.h to 5 then 6 and back to 14 with no effect. 
What should my config.h on the main board look like?
I have a 2-posision switch for SWB, would that matter?
Is there a way of testing the side board separately, (not in dual setup)?

I don't mean to be a pain, but I am really stuck here.

EFeru, 2021-12-18:
Yes, you can flash the Debug variant on sideboard and connect an FTDI adapter to the Serial going to mainboard. Then, use your favorite serial monitor (I used the one from Arduino IDE with baud 115200) to see ASCII data.
At startup a menu should be shown and chose "x" enter to show data from your RC receiver:
https://github.com/EFeru/hoverboard-sideboard-hack-STM/blob/7ed59c6ee29adc99b0dd068e8aec12023320eb22/Src/mpu6050.c#L3764
If you don't see the menu try sending "h" from help.

EFeru, 2021-12-18:
For the main board I would use Variant USART with no additional changes.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/cf7b1d0de1da34612b056c316480f47702a15cbe/Inc/config.h#L311

Then Sideboard goes on short right  cable, while your Serial (Arduino or something) on the long Left cable.

Tjaart-Hobby, 2021-12-18:
Thanks, I will give it a try later today.

EFeru, 2021-12-18:
What MPU sensor do you have on your board? We tested only with MPU6050, other sensors seem to to not work like MPU6050a, we don't know yet, maybe different protocol.

Can you try with MPU sensor disabled? Just comment out this line:
https://github.com/EFeru/hoverboard-sideboard-hack-STM/blob/7ed59c6ee29adc99b0dd068e8aec12023320eb22/Inc/config.h#L47

Tjaart-Hobby, 2021-12-19:
I have a intventsense MPU6050, after I disabled it still, no difference. 
:( and then I messed up the polarity going to the sideboard.      **May it rest in peace.**
I will have to order some sideboards from somewhere, which means weeks of waiting for online buy.
[image]
I have two more sideboards but they are 36v where the others were 16v. It could step 36v down to 16v and use them for both units. I have to google and see what input range the 5v,3v regulators ,on the 36v boards can handle.
[image]
At least the AUX RX/TX on these are already broken out to pads. So I don,t have to solder on to the chip to get at them.
In any case, I wont be able to experiment for a while now and will be flying out to mine site for the next 15 days.

EFeru, 2021-12-19:
Normally you only need one sideboard. Are both of them burned? 
 When you have more info you can post here. Have a good trip!

Tjaart-Hobby, 2022-01-07:
Happy new year
So since last I order a 'new' main-board with two sideboards from e-bay. They arrived with one ST and the other with GD chipped sideboards + again would have to solder leads onto chips. :( 

Luckily just after Christmas and new year there were a few hover-boards up for sale at $50 AUD so grabbed them.
[image]
So on hand I now have x6 GD sideboards and I flashed one as per GD sideboard Git but with SWA on channel 5 for receiver only six channels.
Allso flashed main board as suggested /USART variant/ DUAL with sideboard on right cable.
Still no joy.

[image]
checked USART only > works
              IBUS   only   >works

to do:
tomorrow flash debug in sidebpoard and figure out this FTDI thing I got in the mail to check/test sideboard separate from main-board.
apparently there's a hack using the FTDI to change 6 channels transmitter to 10 channels. I mite give that a go.

Questions: Do I start using the GD sideboard git to open another issue or do I just continue here ? 
Are there any updates or ideas of what could be wrong ?

EFeru, 2022-01-07:
Looks like a nice stock over there, i want the "useless" one 😜

I would say to try flashing your transmitter to 10 channels, normally there shouldn't be a risk there. I've seen many people that successfully have done it.
And then give another try with USART and sideboard Dual input. I would say give a try to also sideboard debug to check that transmitter is working fine with the sideboard.
If you find issues on the GD sideboard you can open an issue there, otherwise we can keep it here as one complete story.

Candas1, 2022-01-07:
Sorry I didn't fully follow the issue, but one thing you could try is to use the web tool instead of the sideboard to make sure at least dual inputs work as expected.
https://github.com/Candas1/Hoverboard-Web-Serial-Control#protocol---hovercar

Candas1, 2022-01-07:
Actually you could even also use the web tool in ibus mode instead of your transmitter/receiver.
https://github.com/Candas1/Hoverboard-Web-Serial-Control#protocol---ibus

It has 4 switches lol, It wasn't used much but I think it should work.
This could help you find out where the problem really is.

Tjaart-Hobby, 2022-01-09:
Update:
Upgraded firmware on remote from 6 Channels to 10

Flashed main board with 'SIDEBOARD_SERIAL_USART2 0' > 
sideboard with receiver works (able to control with RC remote) :( transmitter has to be on before powering bot OR unwanted  behavior.
Flashed main board with '#define CONTROL_SERIAL_USART3' > working (tested with Arduino sketch)
Flashed duel > not working

to do: use web tool to test dual
         Find out how to connect sideboard to laptop with FTDI and view serial monitor. > had a try but unsuccessful

Tjaart-Hobby, 2022-01-17:
Hi still at it :(
I really feel disappointed. I just cant get this dual to work! feels like I have tried everything possible.
tested sideboard serial on both left and rite > OK working 
tested arduino serial > ok working
Dual > not working----- did notice LED s flash diferently almost like the feedback left and rite gets confused.
also found that priorty 1 or 0 seems to clash. Do I add 1/0 after `#define FEEDBACK_SERIAL_USART3  0`
What should my config.h look like?
`// ############################ VARIANT_USART SETTINGS ############################
#ifdef VARIANT_USART
   //#define SIDEBOARD_SERIAL_USART2 0
  //#define CONTROL_SERIAL_USART2  0    // left sensor board cable, disable if ADC or PPM is used! For Arduino control check the hoverSerial.ino
  //#define FEEDBACK_SERIAL_USART2      // left sensor board cable, disable if ADC or PPM is used!

   //#define SIDEBOARD_SERIAL_USART3 0
   #define CONTROL_SERIAL_USART3  0    // right sensor board cable. Number indicates priority for dual-input. Disable if I2C (nunchuk or lcd) is used! For Arduino control check the hoverSerial.ino
   #define FEEDBACK_SERIAL_USART3  0    // right sensor board cable, disable if I2C (nunchuk or lcd) is used!

   #define DUAL_INPUTS                 //  UART*(Primary) + SIDEBOARD(Auxiliary). Uncomment this to use Dual-inputs
  #define PRI_INPUT1             3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #define PRI_INPUT2             3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #ifdef DUAL_INPUTS
    #define FLASH_WRITE_KEY      0x1102  // Flash memory writing key. Change this key to ignore the input calibrations from the flash memory and use the ones in config.h
     #define SIDEBOARD_SERIAL_USART2 1   // left sideboard
     #define FEEDBACK_SERIAL_USART2  1     //TK added
    //#define SIDEBOARD_SERIAL_USART3 1   // right sideboard
    #define AUX_INPUT1           3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT2           3, -1000, 0, 1000, 0     // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #else
    #define FLASH_WRITE_KEY      0x1002  // Flash memory writing key. Change this key to ignore the input calibrations from the flash memory and use the ones in config.h
  #endif

  // #define SUPPORT_BUTTONS_LEFT       // use left sensor board cable for button inputs.  Disable DEBUG_SERIAL_USART2!
  // #define SUPPORT_BUTTONS_RIGHT      // use right sensor board cable for button inputs. Disable DEBUG_SERIAL_USART3!
#endif
// ######################## END OF VARIANT_USART SETTINGS #########################`

Tjaart-Hobby, 2022-01-17:
sorry wrong button.
after thought does the spacing in the code matter? like
..code
...code
not the same or does not compile the write way?

Candas1, 2022-01-17:
I need to use ibus on the sideboard soon. Will let you know if you missed something

Candas1, 2022-01-18:
Hi,

I tried 2 options with usart variant, both worked:
- I enabled dual inputs and used the default settings, so control_usart as primary input on left cable, sideboard_usart as auxiliary input on right cable). I literally just removed // in front of #define dual_inputs
- I also tried you setup, with control_usart as primary input on right cable, and sideboard_usart as auxiliary input on the right cable

Here is the [video](https://www.youtube.com/watch?v=PvN58unZtkc)

Are you sure you have setup the channels properly ? SwA on channel 7 is key to be able to switch to the auxiliary input
[image]

Also try to erase the chip completely in stlink utility before flashing the firmware just in case.

Tjaart-Hobby, 2022-01-18:
Thank you. I will erase the chip tonight and try again. I have my channels
set up like yours. I only have a 2 position switch for SWB and sounds from
the board for all switches but SWA. Is that normal?

Candas1, 2022-01-18:
I also have 2 position switch for SWB.
Emanuel has a more fancy transmitter.

I think that's the problem. The firmware will switch to auxiliary input only if SWA is down. 
Any specific setting on your transmitter like SWA being reversed or something else ?

EFeru, 2022-01-19:
I tested the dual inputs on GD sideboard, the ST version has identical implementation however i did not specifically test this particular dual setup with ST sideboards.

Tjaart-Hobby, 2022-01-19:
YES success ! 
Everything works as you were saying, no need for anything special.
In my case I had to fully erase the chip before re-flashing USART variant with dual control. I am using GD boards as well.

closed 2022-01-19 by Tjaart-Hobby