EFeru/hoverboard-sideboard-hack-STM #11: Mainboard type

holymoly, 2022-07-30:
Does anyone knows this board? I can unlock and download this one but it does not work right now. The Power is only on as long as I hold the button.

[image]

Candas1, 2022-07-30:
Check the main repository's wiki

holymoly, 2022-07-30:
With main repository you mean https://github.com/EFeru/hoverboard-firmware-hack-FOC?
Cause this repo is just linking to itself when clicking on the Wiki button.

Candas1, 2022-07-30:
Yes.
This repository is for sideboards, those have no motor control.

holymoly, 2022-07-30:
Thx for the clarification. So I had a look and downloaded the main Firmware without any luck on the Power supply. After your mentioning of the sideboards and some reading it seems the I got confused with sidboards and splitboards. So as I understood splitboards are the one with the motor control and they are used in master and slave configuration. It seems I need to us the splitboards then. 
But there I can only find firmware for GD.... and not for STM32f103c8t6. Is there one out there?

Candas1, 2022-07-30:
Yes it's mentioned in the compatibility section of the wiki